Rat-OS/RatOS-configuration #39: Allow END_PRINT macro to retract filament so it clears the hot side

rkaronde, 2021-12-31:
Many times after a print is done, and enough time has passed, I wish I could just pull out the filament or use thumbwheel on some extruders to unload the filament, but the filament is "stuck" as the hotend has cooled down.

It would be great if the END_PRINT macro can also retract the filament from the hot parts of the hotend. This could be controlled by another user configured variable.
That way we can still change the filament, if needed, without warming the hotend.

And since the RatOS configuration already knows which hotend is in use, the length to retract could  be added to each hotend cfg file.

miklschmidt, 2022-03-19:
I've been experimenting a bit with this and i've had some pretty awful jamming with PETG as a result. It seems that it's incredibly hard to write an unload macro that avoids string completely, and if we retract the filament completely out of the hot side for easy swapping while cold, there's a very large risk of jamming the hotend if the filament is reloaded (ie. without being swapped) in the next print.

rkaronde, 2022-03-19:
I have used the GCode that Prusa uses for MMU2 unload on another printer to retract from hotside and it seems to work well for BMG-M + Mosquito combo.
Maybe an option in RatOS would be to allow custom script to run before END_PRINT finishes. Between 
`M107`
And
`M117 Done.`

miklschmidt, 2022-03-19:
> I have used the GCode that Prusa uses for MMU2 unload on another printer to retract from hotside and it seems to work well for BMG-M + Mosquito combo. Maybe an option in RatOS would be to allow custom script to run before END_PRINT finishes and before turning hotend heater off

Done: https://github.com/Rat-OS/RatOS-configuration/commit/2a8f40665034f9a452205e7d01e143fa4194f64d

closed 2022-03-19 by miklschmidt