Rat-OS/RatOS-configuration #40: [FR] Allow Z park configurations

psiberfunk, 2022-01-03:
As described inhttps://os.ratrig.com/docs/configuration/macros/  RatOS configuration has park X and park Y parameterss for start/pause/end (ex: *variable_end_print_park_x*), but no "park Z" options.

It would be super useful, particularly for end of prints, I could tell my printer to always park at Z max, or ZMax-10, for example, to make the flex bed i have easier to remove without scraping it against the print nozzle.
l

kmarty, 2022-01-09:
You can follow the comment near "_END_PRINT_AFTER_HEATERS_OFF" macro:
`You can copy these to printer.cfg and modify them to your liking, or just use them as is`

e.g. to:
```
[gcode_macro _END_PRINT_AFTER_HEATERS_OFF]
gcode:
  # Calculate safe Z position
  {% set max_z = printer.toolhead.axis_maximum.z|float %}
  {% set act_z = printer.toolhead.position.z|float %}
  {% if act_z < (max_z - 2.0) %}
      {% set z_safe = [60.0 - act_z,2.0]|max %}
  {% else %}
      {% set z_safe = max_z - act_z %}
  {% endif %}
  # Relative positioning
  G91
  # Retract the filament a bit before lifting the nozzle.
  G1 E-2 F3600
  # Move to safe Z position
  G0 Z{z_safe} F3600
  # Retract filament even more
  G1 E-2 F3600
```
which sufficiently extends the original `at the end of print move Z by 2mm` by `... or to Z=60 if Z < 60`

psiberfunk, 2022-01-09:
Thanks, that helps.. i'm not super-familiar with GCODE macros yet.. it still feels like this value would be nice as a configurable feature of the RatOS distro

psiberfunk, 2022-01-09:
P.S. @kmarty , can you explain the logic of the line  **{% set z_safe = [60.0 - act_z,2.0]|max %}**  ?

I'm not familiar with the syntax in use (I read somewhere it's Jinja2-templating?) enough to know how to parse this into pseudo-code in my head..  but my guess is it means something like:  z_safe = maximum of either 60.0-act_z OR 2.0 ?

kmarty, 2022-01-09:
For example `[1, 4, 3, 5, 2]`  is standard array of values, list. In Jinja you can use "filters" such as `sort`, `min`, `max` _(there are many more, I'm not sure which one of them can be used in Klippers macros. But `min` and `max` definitely can be used)_.
So in this example a `[1, 4, 3, 5, 2]|max` would return the highest value from the list i.e. `5`.

In macro I used `60.0 - act_z` as one value of the list and `2.0` as second value of the list.
If `act_z` would be e.g. 15, then the list would contain `[45.0, 2.0]`. The returned value after `max` filter would be `45.0`.
if `act_z` would be e.g. 75, then the list would contain `[-15.0, 2.0]`. After `max` filter it returns `2.0`.

You don't have to use it here in this way, instead of:
```
{% set z_safe = [60.0 - act_z, 2.0]|max %}
```
 you can use:
```
{% if (60.0 - act_z) > 2.0 %}
  {% set z_safe = 60.0 - act_z %}
{% else %}
  {% set z_safe = 2.0 %}
{% endif %}
```
I just want to have this way because with using `max` filter, only one line is changed in comparing to the "default" macro.

miklschmidt, 2022-09-26:
Fixed in `development`

closed 2022-09-26 by miklschmidt