Rat-OS/RatOS-configuration #42: FR: Proper newline at the end of of cfg any files

kmarty, 2022-01-05:
Alhough nobody should edit cfg files provided by this repository, proper newline at the end of file (which is done automatically in any unix system) would be really appreciated.
If nothing, at least to avoid unnecessary diffs like:
```
--- pp/24v-0.7a-e.cfg	2022-01-05 19:32:14.000000000 +0100
+++ p/24v-0.7a-e.cfg	2022-01-05 19:33:00.000000000 +0100
@@ -6,4 +6,4 @@
 driver_TBL: 1
 driver_TOFF: 3
 driver_HEND: 9
-driver_HSTRT: 7
\ No newline at end of file
+driver_HSTRT: 7
```
where literaly no config change was made. The very same "error" is shown e.g. here:
[image]

In short, non-binary file shouldn't exists without proper newline at the end of file.

Thank you very much.

kmarty, 2022-01-06:
If someone is curious why it is bad, why (text) file with no eol at eof is ... corrupted, here is an example (one of many):
Let's say we have two files.
file_a:
```
first line of A
second line of A
third line of A
```
and file_b:
```
first line of B
second line of B
third line of B
```

If "file_a" wouldnt' have newline at the end of file, their concatenation would be like this:
```
first line of A
second line of A
third line of Afirst line of B
second line of B
third line of B
```
instead of this:
```
first line of A
second line of A
third line of A
first line of B
second line of B
third line of B
```

miklschmidt, 2022-02-18:
Absolutely, i usually always have new lines at the end of my files, apparently i forgot to check the box in my home vscode setup (and now i wonder why this isn't the default...). Will get this fixed.

miklschmidt, 2022-02-18:
Done.

closed 2022-02-18 by miklschmidt