Rat-OS/RatOS-configuration #53: Change Homing Behavior For GENERATE_SHAPER_GRAPHS

brenond, 2022-02-19:
Hi Mikkel, as per your request I've moved this issue over here.  Thanks for looking into it!

Some users have detachable accelerometers that mount directly to the printer's hotend. Depending on the setup, this can end up closer to the bed than the ABL probe on the carriage, causing a crash if the printer homes after installing the accelerometer.

Currently, even if you have homed the printer beforehand and the printhead's location is known, the printer will still home all axes when the GENERATE_SHAPER_GRAPHS command is run, potentially causing a crash or cable snag.

This is not urgent, but would be a nice change when time allows.

miklschmidt, 2022-03-19:
Thank you for the suggestion! 
Fixed in https://github.com/Rat-OS/RatOS-configuration/commit/3fe9997953ce8561e6b9bca39248435cd45aef39

miklschmidt, 2022-03-19:
So that fix broke [homing_override], this will be delayed a bit. Reopening the issue.

miklschmidt, 2022-03-19:
Fix implemented via MAYBE_HOME macro :)

closed 2022-03-19 by miklschmidt