Rat-OS/RatOS-configuration #55: Issue with Sensorless Homing on new config

Anon0ne, 2022-03-19:
I updated my configurations, and after doing so my machine failed to home. I was doing screenshots for a review I am writing; but I realized that no matter which way I modified your new script it would not home.

The first axis will home, and then the second one will not even attempt. It will then immediately try to go to half of the max distance, and home the Z axis which results in crashing etc. I reverted to the old configuration and my system will home once again.

asierralozano, 2022-04-30:
+1 here. It is happening on the VCore3 sensorless homing config.

miklschmidt, 2022-09-26:
This is because the current sensorless homing template is optimized for the Minion. You're gonna need to adjust your currents before homing Y on the V-Core 3. I do not recommend sensorless homing on a CoreXY, there are only downsides compared to endstops. It's not worth it.

closed 2022-09-26 by miklschmidt