Rat-OS/RatOS-configuration #98: homing_override 'middle'

JiMcNulty, 2023-01-22:
homing_override 'middle' doesn't take into account the probe location, so the z homed middle is not in the middle of bed

miklschmidt, 2023-01-28:
It shouldn't be. Middle = nozzle position, probe offsets will take care of the rest.

closed 2023-01-28 by miklschmidt

JiMcNulty, 2023-01-28:
Dude...
I wouldn't go to the trouble of submitting this if there wasn't a reason
probe offsets is not applied in this case, and in this case only.
Please verify that when you do G28 Z your PROBE is in the middle of the bed (hint - it doesn't) (unless something has changed in v2.0)

JiMcNulty, 2023-01-28:
The fastest way to check it is by doing g28 z

[image]

If it's at 250, 250 - your probe is not in the middle

miklschmidt, 2023-01-28:
Maybe i didn't make it clear, but the probe offset is applied internally everywhere it matters. Please correct me if i'm wrong, but i don't see any point in centering the probe on the bed during homing, It doesn't do anything useful, it's just added complexity.

JiMcNulty, 2023-01-28:
Benefits:
1. Improve the stability of the z offset
2. Improve the stability of the z-tilt (when you tilt a plan using Maxwell coupling the middle point shouldn't change)

Just things that I discovered while working on FDC, this little thing was a make or break for general stability of offsets, bed meshes and z tilt

I don't see the complexity that it adds 
Also and correct me if i'm wrong, when doing native Klipper G28 Z without the Ratos home override it does take probe offset into account

miklschmidt, 2023-01-28:
1 and 2) That's not entirely correct. It's not a prerequisite for a maxwell coupling that the pivot point is in the center, and it isn't on the V-Core 3, for example. Furthermore it depends on the beds topography anyway. We home after Z-tilt regardless, so that's a non issue to begin with.

3) Also a bit of a non issue, but then we should aim for that and not just subtract probe offsets. I don't think the current mesh is centered on the bed in all cases, and what if the user changes their mesh min/max?

> Also and correct me if i'm wrong, when doing native Klipper G28 Z without the Ratos home override it does take probe offset into account
I actually don't know. This is your best argument, if that's what klipper does natively, then we should do it to.

miklschmidt, 2023-01-28:
Maybe it's better to home close to one of the z_tilt pivot points instead. Those spots should be the most stable.

JiMcNulty, 2023-01-28:
As you wish...don't see the big deal in fixing it
It can only help
There is no down side 
It's a simple subtraction, just one line of code

miklschmidt, 2023-01-28:
> As you wish...don't see the big deal in fixing it It can only help There is no down side It's a simple subtraction, just one line of code

I need to be convinced that it's actually a "fix", each change comes with a risk of breaking in some edge case nobody thought of.

miklschmidt, 2023-01-28:
Actually, this change would break for bltouch users, so there's that.

JiMcNulty, 2023-01-29:
> Actually, this change would break for bltouch users, so there's that.

That's a good point, good CR

[#](https://github.com/JiMcNulty/RatOS-configuration/pull/1#issue-1561056309)

JiMcNulty, 2023-01-29:
> > As you wish...don't see the big deal in fixing it It can only help There is no down side It's a simple subtraction, just one line of code
> 
> I need to be convinced that it's actually a "fix", each change comes with a risk of breaking in some edge case nobody thought of, so there has to be an upside to justify that risk.

That's true 
Anyway, updated the PR the make the decision easier

Just trying to help ;)