Rat-OS/RatOS-configuration #102: ADXL345 axis mapping

GP3D84, 2023-01-30:
Hi, according to official klipper documentation the accelerometer must be installed respecting correct axis
[image]
On classic Eva3 mounting, the adxl axis should be remapped by : axes_map: x,  y, z (default) to axes_map: x, z, y in RatOS.
In any case is better to warning user about alternative adxl mount

@miklschmidt  what do you think?

miklschmidt, 2023-01-31:
@miklschmidt AFAIK this does little more than make sure your graph has the correct legends. There are many, many ways to mount ADXL345's currently, and this is something the user can easily change on their own. I don't think the work and complexity in making sure the axes are mapped correctly in all use cases is worth it.

GP3D84, 2023-01-31:
The legends is always correct despite the incorrect mapping.
What changes is the correct reading.
Not only the klipper documentation say to respect axis vs adxl mount but in other applications of adxl as well.
Anyway, if you don't think it's important, i will close this issue
Thank you anyway for all the work you've done

miklschmidt, 2023-01-31:
> The legends is always correct despite the incorrect mapping.

@GP3D84 What i mean is that it'll show Y vibrations for X etc, depending on the orientation.

> Not only the klipper documentation say to respect axis vs adxl mount

Can you point me to where it says that?