Rat-OS/RatOS-configuration #126: Error executing manual mesh with Euclid probe, using profile name with a space

etotheipi, 2023-04-27:
### What happened

I tried to generate a mesh from the "Heightmap" menu after updating to a Euclid probe, and used the name "Euclid Mesh" which then threw an error that indicates the name is not being quoted properly:

Malformed command `'BED_MESH_CALIBRATE_ORIG PROFILE=Euclid Mesh'`

When I rerun with a name that has no space, it works fine.

### What did you expect to happen

I expected it to properly encapsulate the profile name with quotes so it doesn't treat the user-provided name as more than one argument.

### How to reproduce

Do the stowable-probes thing and then try to generate a mesh from the "HEIGHTMAP" menu using a name that has a space in it.

miklschmidt, 2023-05-15:
Will be fixed once #137 is merged

miklschmidt, 2024-02-14:
fixed a year ago, apparently forgot to close :D

closed 2024-02-14 by miklschmidt