Rat-OS/RatOS-configuration #144: Slow UI load times

jLynx, 2023-07-17:
### What happened

When I load the webpage it takes around 8-20 seconds to load. (Validated on multiple high-end devices)

### What did you expect to happen

The page to load a lot faster than it currently does

### How to reproduce

Im not sure. Its a stock config. just with a lot of use time (438hours)

### Additional information

https://github.com/Rat-OS/RatOS-configuration/assets/4393979/61c7380c-8376-4ade-ab3c-a7b693c39aa6

https://github.com/Rat-OS/RatOS-configuration/assets/4393979/4aa5a27d-b621-418f-bfd6-3ccf810838ad

jLynx, 2023-07-17:
Apparently it could be to do with the file history, so I have created this too https://github.com/mainsail-crew/mainsail/issues/1468

miklschmidt, 2023-07-25:
I'd argue that loading stuff before it's needed is a rookie web development mistake.

miklschmidt, 2023-10-22:
@jLynx i'm closing this as it's a mainsail performance issue. Feel free to reopen if you have any questions.

jLynx, 2023-10-22:
Sounds good to me

closed 2023-10-22 by jLynx