Rat-OS/RatOS-configuration #153: _START_PRINT_AFTER_HEATING_BED cools to preheat_extruder_temp

SirUli, 2023-10-23:
### What happened

When the extruder is already heated over the preheat_extruder_temp temperature, the Macro _START_PRINT_AFTER_HEATING_BED waits until the extruder is cooled down to the pre-heat temperature and then only proceeds (basically to heat up again in my case). See [here](https://github.com/Rat-OS/RatOS-configuration/blob/d0656945750c36f76fb547665309f810a6e0c46e/macros.cfg#L436-L447)

### What did you expect to happen

When the extruder temperature is already higher than ``preheat_extruder_temp``, the Macros could be skipped entirely.

### How to reproduce

Heat extruder over preheat_extruder_temp, then start a print. You will notice that the temperature will drop first while the macros is being awaited. See line at the top for this - which first drops to preheat_extruder_temp and then goes up again.

[image]

miklschmidt, 2023-11-14:
That is intended for a couple reasons:
1) it's a bandaid for temperature dependent probes.
2) It prevents oozing during z-tilt and bed meshing.
3) The extra time helps for heat stabilization (though it's not enough by it's own).

If you want to change this behavior, you can override _START_PRINT_AFTER_HEATING_BED to use `M104` and `TEMPERATURE_WAIT` with a `MIN` parameter instead.

closed 2023-11-14 by miklschmidt

smwoodward, 2023-11-15:
> That is intended for a couple reasons:
>
> 1. it's a bandaid for temperature dependent probes.
> 2. It prevents oozing during z-tilt and bed meshing.
> 3. The extra time helps for heat stabilization [...]

Something that could be done here is have the print start macro check if a temp dependent probe is configured and only drop the temp for those probes.  That wouldn't necessarily fix the bed meshing and z-tilt, but maybe it could be a check if the user wants z-tilt and meshed before printing.

miklschmidt, 2023-11-15:
@smwoodward i'm trying to keep RatOS to best practices, and not overcomplicate things for small edge cases or different preferences, but still provide an escape hatch to easily facilitate such things. Currently i'm OK with overriding this fairly specific macro for situations like these.