Rat-OS/RatOS-configuration #157: Manta M8P V2.0

trudeau87, 2023-12-17:
### Is your feature request related to a problem? Please describe

No option for Manta M8P V2.0

### Describe the solution you'd like

Can this be added to the config of the RatOS?

sergelhr, 2024-01-05:
> ### Is your feature request related to a problem? Please describe
> 
> No option for Manta M8P V2.0
> ### Describe the solution you'd like
> 
> Can this be added to the config of the RatOS?
> ### Describe alternatives you've considered
> 
> _No response_
> ### Additional information
> 
> _No response_

I'm using this board already see https://github.com/sergelhr/RatOS-configuration

miklschmidt, 2024-02-14:
This is implemented in the `development` branch and will be published with 2.1

miklschmidt, 2024-02-17:
@LivioB74 Unfortunately i can't give a specific date at this point, but we're getting close to public beta. There are some major features being developed and tested right now that need to be completed first. 

That said, if you can live with the update manager report "dirty" in mainsail, you can just simply install RatOS 2.0, update all packages, then dump [this folder](https://github.com/Rat-OS/RatOS-configuration/tree/v2.x/boards/btt-manta-m8p) into `/home/pi/printer_data/config/RatOS/boards`. Then ssh into the pi and run the ratos update script:
```bash
ssh pi@ratos.local
sudo ./printer_data/config/RatOS/scripts/ratos-update.sh
```

And then go to the configurator to flash your Manta M8P v2.0. To use the board in printer.cfg, comment out the existing boards and add `[include RatOS/boards/btt-manta-m8p-20/config.cfg]`. Now your Manta M8P v2 works in RatOS v2.0. You just won't be able to update RatOS without resetting it or you'd have to drop it back in again.

miklschmidt, 2024-02-17:
Not quite how it goes. You know [we have documentation](https://os.ratrig.com/docs/installation) for the setup process right? Just follow that. Before you go to boards (so after configuring wifi) update everything through mainsail/moonraker ( i'm not talking about the system packages), dump the folder in and run the ratos-update.sh script. Then proceed with the guide at `/configure`, and everything should be the same - except you have to add the board include yourself.

miklschmidt, 2024-02-17:
@LivioB74 No, that's the manta m8p v1 from the v2.x branch. You want this folder https://github.com/Rat-OS/RatOS-configuration/tree/development/boards/btt-manta-m8p-20

It's the same repository, you're just replicating the files for that board in your local v2.x copy of the repository. Same folder name, same file contents.

It's complaining because you duplicated a board config (the manta m8p v1).

IridianZore, 2024-04-05:
Hi,
Do you have any update when the release of 2.1 will be?

miklschmidt, 2024-04-06:
@IridianZore it will coincide with the release of the V-Core 4 :)

IridianZore, 2024-04-07:
I copied the files on the above link to the boards file of the OS.
Then I started RatOS Setup Wizard. 
Unfortunately I always get the same error.
Invalid board definition(s) in /home/pi/printer_data/config/RatOS/boards.

Can you help me get my manta m8p v2 with cb1 working.

IridianZore, 2024-04-07:
> I'm using this board already see https://github.com/sergelhr/RatOS-configuration

Can you give me a workflow how to handle it.
I'm a newbe and don't have this deep knowledge....

miklschmidt, 2024-04-07:
> I copied the files on the above link to the boards file of the OS. Then I started RatOS Setup Wizard. Unfortunately I always get the same error. Invalid board definition(s) in /home/pi/printer_data/config/RatOS/boards.
> 
> Can you help me get my manta m8p v2 with cb1 working.

Ah yes, i forgot serialPath is removed in 2.1.

try adding `"serialPath": "/dev/btt-manta-m8p-20",` after the first line (with the opening `{`) to `boards/btt-manta-m8p-20/board-definition.json`

IridianZore, 2024-04-07:
you mean as below mention?
"dfu": {
		"serialPath": "/dev/btt-manta-m8p-20"
		"flashDevice": "0483:df11",
		"dfuBootImage": "dfubooting.png",
		"instructions": [
			"Make sure the board is powered on",
			"Press and hold the BOOT 0 button on the board.",
			"Press and release the RESET button on the board.",
			"Release the BOOT 0 button."
		],
		"hasBoot0Jumper": false
	},

IridianZore, 2024-04-07:
[image]

miklschmidt, 2024-04-07:
> you mean as below mention? "dfu": { "serialPath": "/dev/btt-manta-m8p-20" "flashDevice": "0483:df11", "dfuBootImage": "dfubooting.png", "instructions": [ "Make sure the board is powered on", "Press [...]

No if that's the contents of your board-definition.json then you didn't copy the files or you edited and deleted some of the board definition file. It should be inserted here [below this line in the file](https://github.com/Rat-OS/RatOS-configuration/blob/development/boards/btt-manta-m8p-20/board-definition.json#L3).

IridianZore, 2024-04-07:
I just picked out the part with the entry.
This is the complet fil
[board-definition.json](https://github.com/Rat-OS/RatOS-configuration/files/14899305/board-definition.json)

miklschmidt, 2024-04-07:
> I just picked out the part with the entry. This is the complet fil [board-definition.json](https://github.com/Rat-OS/RatOS-configuration/files/14899305/board-definition.json)

Gotcha. Anyway, it should not go under "dfu", it needs to be at the top level, it also needs a comma at the end. This is what you want:
```json
{
	"$schema": "../board-definition.schema.json",
	"serialPath": "/dev/btt-manta-m8p-20",
	"name": "Manta M8P v2.0",
	"manufacturer": "BIGTREETECH",
	"id": "btt-manta-m8p-20",
	"firmwareBinaryName": "firmware-btt-manta-m8p-20.bin",
	"compileScript": "compile.sh",
	"flashScript": "flash.sh",
	"documentationLink": "https://os.ratrig.com/docs/boards/btt/manta-m8p-11",
	"fourPinFanConnectorCount": 2,
	"driverCount": 8,
	"driverVoltages": [24, 36, 48, 56],
	"dfu": {
		"flashDevice": "0483:df11",
		"dfuBootImage": "dfubooting.png",
		"instructions": [
			"Make sure the board is powered on",
			"Press and hold the BOOT 0 button on the board.",
			"Press and release the RESET button on the board.",
			"Release the BOOT 0 button."
		],
		"hasBoot0Jumper": false
	},
	"motorSlots": {
		"MOTOR1": {
			"title": "MOTOR 1",
			"enable_pin": "PC14",
			"step_pin": "PE6",
			"dir_pin": "PE5",
			"uart_pin": "PC13",
			"cs_pin": "PC13",
			"diag_pin": "PF4",
			"endstop_pin": "PF4",
			"spi_software_mosi_pin": "PC12",
			"spi_software_miso_pin": "PC11",
			"spi_software_sclk_pin": "PC10"
		},
		"MOTOR2": {
			"title": "MOTOR 2",
			"enable_pin": "PE4",
			"step_pin": "PE2",
			"dir_pin": "PE1",
			"uart_pin": "PE3",
			"cs_pin": "PE3",
			"diag_pin": "PF3",
			"endstop_pin": "PF3",
			"spi_software_mosi_pin": "PC12",
			"spi_software_miso_pin": "PC11",
			"spi_software_sclk_pin": "PC10"
		},
		"MOTOR3": {
			"title": "MOTOR 3",
			"enable_pin": "PE0",
			"step_pin": "PB8",
			"dir_pin": "PB7",
			"uart_pin": "PB9",
			"cs_pin": "PB9",
			"diag_pin": "PF2",
			"endstop_pin": "PF2",
			"spi_software_mosi_pin": "PC12",
			"spi_software_miso_pin": "PC11",
			"spi_software_sclk_pin": "PC10"
		},
		"MOTOR4": {
			"title": "MOTOR 4",
			"enable_pin": "PB6",
			"step_pin": "PB4",
			"dir_pin": "PB3",
			"uart_pin": "PB5",
			"cs_pin": "PB5",
			"diag_pin": "PF1",
			"endstop_pin": "PF1",
			"spi_software_mosi_pin": "PC12",
			"spi_software_miso_pin": "PC11",
			"spi_software_sclk_pin": "PC10"
		},
		"MOTOR5": {
			"title": "MOTOR 5",
			"enable_pin": "PG15",
			"step_pin": "PG13",
			"dir_pin": "PG12",
			"uart_pin": "PG14",
			"cs_pin": "PG14",
			"diag_pin": "PF0",
			"endstop_pin": "PF0",
			"spi_software_mosi_pin": "PC12",
			"spi_software_miso_pin": "PC11",
			"spi_software_sclk_pin": "PC10"
		},
		"MOTOR6": {
			"title": "MOTOR 6",
			"enable_pin": "PG11",
			"step_pin": "PG9",
			"dir_pin": "PD7",
			"uart_pin": "PG10",
			"cs_pin": "PG10",
			"diag_pin": "PC15",
			"endstop_pin": "PC15",
			"spi_software_mosi_pin": "PC12",
			"spi_software_miso_pin": "PC11",
			"spi_software_sclk_pin": "PC10"
		},
		"MOTOR7": {
			"title": "MOTOR 7",
			"enable_pin": "PD6",
			"step_pin": "PD4",
			"dir_pin": "PD3",
			"uart_pin": "PD5",
			"cs_pin": "PD5",
			"spi_software_mosi_pin": "PC12",
			"spi_software_miso_pin": "PC11",
			"spi_software_sclk_pin": "PC10"
		},
		"MOTOR8": {
			"title": "MOTOR 8",
			"enable_pin": "PD2",
			"step_pin": "PC7",
			"dir_pin": "PC8",
			"uart_pin": "PC6",
			"cs_pin": "PC6",
			"spi_software_mosi_pin": "PC12",
			"spi_software_miso_pin": "PC11",
			"spi_software_sclk_pin": "PC10"
		}
	},
	"ADXL345SPI": {
		"cs_pin": "PB12",
		"software": {
			"sclk": "PC10",
			"mosi": "PC12",
			"miso": "PC11"
		}
	},
	"stepperSPI": {
		"software": {
			"sclk": "PC10",
			"mosi": "PC12",
			"miso": "PC11"
		}	
	}
}
```

IridianZore, 2024-04-08:
> -

YES! ;) It worked! Great job! Thank you so much!
[image]

aardvarkglory, 2024-05-28:
I followed this as mentioned above.  I am able to click on the M8P v2 board and it goes to the next page.  It gives me 2 options for flashing, DFU or Manual SD.  When I click on the button for DFU or Manual flash, the button itself quickly flashes and does nothing.  Is there anything I can do to make this feature work?

sergelhr, 2024-05-30:
Hello did you try to flash the bootloader ?
https://github.com/bigtreetech/Manta-M8P/tree/master/V2.0/Firmware

miklschmidt, 2024-06-01:
v2.1.0-RC1 has been released with native support.

closed 2024-06-01 by miklschmidt

aardvarkglory, 2024-06-01:
> v2.1.0-RC1 has been released with native support.

Thanks and awesome work on 2.1.  Installed it and fixed this issue among other issues.  Thanks.