Rat-OS/RatOS-configuration #272: Prusa MK3S generated configuration errors

david-antos, 2024-06-14:
### What happened

On RatOS 2.1, I have generated configuration for Prusa MK3S with Einsy board with the following parameters in the wizard:
[image]

It has following issues (I've repaired them step by step in the config manually in order to get complete list at once)

1. Option 'points' in section 'quad_gantry_level' must be specified (quad_gantry_level entry in the config is quite pointless there, it can be corrected by removing the entry completely)

2. tmc virtual endstop requires diag pin config
corrected by adding

```
[tmc2130 stepper_x]
...
diag1_pin: !x_diag_pin
...
```
etc also for y, z, e
(note the !, otherwise the toolhead crashes violently)

3. MCU temperature not supported on atmega2560 (necessary to get rid of the [teperature_sensor Einsy] section)

4. Pin 'display_rs_pin' is not a valid pin name on mcu 'mcu' (none of the display pins are defined for Einsy)

5. Pin 'filament_switch_sensor_pin' is not a valid pin name on mcu 'mcu' (again, missing definition of pin names)

### What did you expect to happen

The wizard should produce a valid and functional configuration

### How to reproduce

As described above

### Additional information

The pins are defined in the RatOS/boards/prusa-einsy/config.cfg, that file is not included in the config (I suppose that's the purpose of RatOS 2.1)

3dprintpt, 2024-08-08:
Thanks David.
I will try to reproduce this. No MK3S but I should be able to look at the files and check what is missing.
Will see ;)

david-antos, 2025-01-03:
MK3S configuration has been tuned in the development version of 2.1 recently and verified to be working, making this issue irrelevant.

closed 2025-01-03 by david-antos