Rat-OS/RatOS-configuration #275: Hotend / Extruder configuration not working for PT1000 sensor

romarysonrier, 2024-06-26:
### What happened

Currently the only settings available extruder temperature sensor is "Hotendthermistor".

But it doesn't work when using PT1000, as it is not setting the proper klipper configuration.

On Octopus Pro, for [extruder] , It is setting :

heater_pin: PA0
sensor_type: PT1000
sensor_pin: PF4
pullup_resistor: 3700

what is not correct.

what's more "pullup_resistor:3700" prevent any override in printer.cfg because klipper doesn't support this setting for MAX31865

### What did you expect to happen

On Octopus Pro, for [extruder], It should set :

sensor_type: MAX31865
sensor_pin: PF8
spi_software_sclk_pin: PA5
spi_software_mosi_pin: PA7
spi_software_miso_pin: PA6
spi_speed: 4000000
rtd_nominal_r: 1000
rtd_reference_r: 4300
rtd_num_of_wires: 2
rtd_use_50Hz_filter: True

### How to reproduce

Vcore 3
Octopus Pro V1.1
All latest version

### Additional information

I would suggest to add in the UI a hotend tempreratur sensor selectbox , to allow the user to chosse the proper sensor chip. And then it would allow to restrict the current Thermistor select box only to valid values for the selected chip.

kmarty, 2024-06-27:
Connecting PT1000 to TH0 pin, i.e. the same way as NTC thermistor, is perfectly valid option. The `pullup_resistor` might be off, since on TH0 is 4.7k pull-up.
And `PA0` as heater_pin doesn't sound either (Octopus Pro v1.0 has `PA2`, `PA3`, `PB10` and `PB11` as heater pins)
But the part:
```
sensor_type: PT1000
sensor_pin: PF4
```
is ok. It is one of the options how to use PT1000  (not PT100, which unconditionally requires MAX chip)

HelgeKeck, 2024-06-27:
the max chip creates mostly issues with klipper for no real benefit for the user. jsut connect it according to the ratos documentation to the normal th connector and it will work just fine

HelgeKeck, 2024-06-27:
only thing you have to do in ratos 2.1 is to select PT1000 in the configurator and it will work. in some cases it even tells you to set a pull up resistor jumper to improve the accuracy of it

miklschmidt, 2024-09-27:
What they guys said is correct. However if you want to use a MAX31865 instead you can do that by picking any other thermistor and configuring it yourself in printer.cfg like regular klipper :)

There are no plans to support it directly in the configurator for now, as it's often a huge point of struggle for people to use with no discernible benefit (for PT1000's).