Rat-OS/RatOS-configuration #324: MCU Protocol error

igor7205, 2024-12-22:
### What happened

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-396-gb7233d119-dirty
MCU(s) which should be updated:
mcu: Current version v0.12.0-396-gb7233d119
rpi: Current version v0.12.0-396-gb7233d119
btt_lis2dw: Current version v0.12.0-114-ga77d0790
Up-to-date MCU(s):
<none>

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'btt_lis2dw': Unable to extract params from: config_lis2dw

### What did you expect to happen

If I turn off the config then everything works,

### How to reproduce

I also have the config in the overwrite section, somehow the system wants the same clipper version from all mcu

GP3D84, 2024-12-22:
this has nothing to do with ratos
you need to flash your beacon clone  to the latest firmware version.

igor7205, 2024-12-22:
This is an adxl connected to usb on the pi, no beacon, has always worked well until today