Rat-OS/RatOS-configuration #331: Doubled Z-offset after running macros load_filament and then print file

RSEvseev, 2025-03-05:
### What happened

Hi!
Ratrig Wideboy 3.1, z-probe on right toolhead.
Offset of left toolhead is 0.65mm
When I run macros load_filament and then run print (left toolhead), i saw z-offset is 1.3mm after priming (and 0.65mm before priming).
When I correct idex_zoffset to 0.33 and run load_filament and then start printing, i saw z-offset is 0.66 after printing prime blob.
Same behaviour with negative values of z-offset.
No any logs of z-offset changing in console and klippy.log
After restart RatOS and printing without load_filament, all fine, no more doubling z-offset.
Bug is repeatable after each filament loading.

### What did you expect to happen

I expect correct z-offset after filament loading

### How to reproduce

1. Idex, default toolhead 1 (z-probe on right toolhead)
2. set z-offset of left toolhead is not equal 0 (less then 0 for safe reproducing)
3. run load_filament macro on toolhead 0
4. run print (left toolhead), waiting for print prime blob
5. see z-offset value in web-interface

miklschmidt, 2025-03-05:
@RSEvseev what version of RatOS are you using, and how are you adjusting the z-offset in step 2?

RSEvseev, 2025-03-06:
@miklschmidt 
RatOS Core: Configurator & Configuration v2.1.0-RC3-26-g33980992 (latest at this moment)
Adjust default (right) toolhead by probe_calibrate with paper, then adjust other toolhead by set z to 1mm and decrease z by 0.1mm step-by-step with paper check. After, I correct idex_zoffset variable in ratos-variables.cfg

HelgeKeck, 2025-03-06:
> [@miklschmidt](https://github.com/miklschmidt) RatOS Core: Configurator & Configuration v2.1.0-RC3-26-g33980992 (latest at this moment) Adjust default (right) toolhead by probe_calibrate with paper, [...]

you may jsut have accidentaly inverted the z-offset in the variabes file

RSEvseev, 2025-03-06:
> you may jsut have accidentaly inverted the z-offset in the variabes file

No, I print with this value, the print is excellent (before running the load_filament macro). 
Other values were to check the repeatability of the bug. But after running load_filament macro in first print, z-offset still correct on making prime blob, but automatically doubled after it, on printing model. Second print or print after reboot is ok.
I can make screen capture of web-interface in process for understanding.

RSEvseev, 2025-03-06:
Video with bug reproduce
https://youtu.be/_wSuW256Ovs?si=f2owUUjxOki5Jyz4
Load filament, first print with bug, then next print without bug
Bug in 3:40-4:00 interval
No bug on 6:40-7:00 interval