Rat-OS/RatOS-configuration #332: Fresh install of 2.02 on Vcore3 "Must home axis-first"

sevenmead, 2025-03-09:
### What happened

I have had this machine running flawlessly for 2 years ish and yesterday due to a really simple mistake I am now on a fresh install.

I am currently setting the bltouch probe offset, every time I save a setting in config and reboot, the machine reboots to a state where it it can't home

"Must home axis first: 0.000 0.000 15.000 [0.000]"

I have to set kinematic center, then it will home.

It does this every time it reboots. 

There are a decent amount of reports of this but most are 2 years old. And most of those were resolved with turning off the steppers. I could only find one example of someone having something similar. 

I have only modified the config.cfg very lightly to reflect my machine setup.

### What did you expect to happen

to home.

### How to reproduce

install fresh 2.02 on an SD card, run through the wizard, change the config file to reflect 500mm, bltouch, and rapido hotend

sevenmead, 2025-03-09:
Now something real strange is happening. When I Save&restart and the KLIPPER: REPORTS SHUT DOWN screen is up, it offers you the option to restart or firware restarte and clicking them does nothing. using the dropdown power menu the command gets sent but still nothing happens. having to plug and unplug at this point.

pkucmus, 2025-10-19:
Hi, I have the same thing happening after bumping my VC3.1 to the newest stable from today: 

[image]

The ugly workaround which I'm sure @miklschmidt will scold me for but it works for now is:

```
# printer.cfg

[ratos_homing]
axes: 
z_hop: 0

[safe_z_home]
home_xy_position: 150,150
z_hop: 10
z_hop_speed: 30.0
```